Clon1998/mobileraker_companion #96: Errors in Docker instance of companion

pwlgrzs, 2024-06-11:
Hi,

I noticed some errors in the log of my docker instance, I was looking at them trying to find out why I am getting no notifications:

`2024-06-11 16:35:51 asyncio ERROR Task exception was never retrieved

future: <Task finished name='Task-205133' coro=<MobilerakerCompanion._evaluate_with_timeout() done, defined at /opt/mobileraker/mobileraker/mobileraker_companion.py:74> exception=AttributeError('No language-entry found for "%s"', 'state_started_body')>

Traceback (most recent call last):

  File "/opt/mobileraker/mobileraker/mobileraker_companion.py", line 85, in _evaluate_with_timeout

    await asyncio.wait_for(self._evaluate(snapshot), timeout=60)

  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for

    return await fut

           ^^^^^^^^^

  File "/opt/mobileraker/mobileraker/mobileraker_companion.py", line 115, in _evaluate

    state_noti = self._state_notification(cfg, snapshot)

                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/opt/mobileraker/mobileraker/mobileraker_companion.py", line 343, in _state_notification

    body = translate_replace_placeholders(

           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/opt/mobileraker/mobileraker/util/i18n.py", line 144, in translate_replace_placeholders

    translation = translate(companion_config.language, str_key)

                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

  File "/opt/mobileraker/mobileraker/util/i18n.py", line 134, in translate

    raise AttributeError('No language-entry found for "%s"', str_key)

AttributeError: ('No language-entry found for "%s"', 'state_started_body')`

Please let me know if you wish for a full log.

Clon1998, 2024-06-13:
Hey,
thanks for reporting the issue.
I will have a look to see how to resolve that issue.

pwlgrzs, 2024-06-13:
Here's my file:
```
[general]
language: en 
timezone: Europe/Warsaw 
eta_format: %%d.%%m.%%Y, %%H:%%M:%%S
include_snapshot: True

[printer Ender3V3SE]
moonraker_uri: ws://192.168.88.171:7125/websocket
moonraker_api_key: False
snapshot_uri: http://192.168.88.171/webcam/?action=snapshot

[printer Voron]
moonraker_uri: ws://192.168.88.172:7125/websocket
moonraker_api_key: False
snapshot_uri: http://192.168.88.172/webcam/?action=snapshot
```

I've been running this config without any changes for a long time.

Clon1998, 2024-06-14:
I fixed the issue by using the correct i18n/translation key for the print start notification.

closed 2024-06-14 by Clon1998

pwlgrzs, 2024-06-14:
> I fixed the issue by using the correct i18n/translation key for the print start notification.

Thanks, all is working again!

commit 9968ffbc (2024-06-14): Fix #96 - Usage of wrong language key for print start notifcations