Clon1998/mobileraker_companion #98: ETA in notification is wrong timezone

dragoonmc, 2024-07-25:
When I setup Mobileraker and the companion I had the timezone on my printer set to default instead of EST. I fixed the printers timezone but still get ETA that is 5 hours off in Mobil notification

Clon1998, 2024-07-25:
Hey,
did you change the timezone in the `mobileraker.conf` file too?

```python
[general]
language: en 
# !!! DEPRECATED. The app now syncs the app's language to the companion
# one of the supported languages defined in i18n.py#languages (de,en,...)
# !!! For users from the UK: entering 'uk' will resolve to Ukrainian language, not English. Use 'en' for English!
# Default: en
timezone: Europe/Berlin 
# correct timezone e.g. Europe/Berlin for Berlin time or US/Central. 
# For more values see https://gist.github.com/heyalexej/8bf688fd67d7199be4a1682b3eec7568
# Default: Tries to use system timezone
# Optional
```

dragoonmc, 2024-07-25:
no I did not lol I expected that default to following system time. done and looking right now
thanks

closed 2024-07-25 by dragoonmc

Clon1998, 2024-07-26:
I think I had to implement it that way because of a limitation some users encountered