Clon1998/mobileraker_companion #103: Elegoo Neptune 4 Max

ejdprops, 2024-11-26:
Before I take the time and effort to try and install the companion, can anyone tell me if the Debian install will work with the Elegoo Neptune 4 Max version of Klipper?

Clon1998, 2024-11-26:
I don't own a Neptune printer but it works fine on my Debian distribution. In any way installation via KIAUH or the installer simplifies the installation.

MisterPl0w, 2024-12-01:
I own the Neptune 4 Plus, which has the same brains as the max. The install script worked flawlessly and the app almost immediately picked up the first notification (that the printer was in standby). So far so great!

I think this issue can be closed.

Clon1998, 2024-12-01:
@MisterPl0w thanks for letting me know.
I'll close the issue in that case.

closed 2024-12-01 by Clon1998