Clon1998/mobileraker_companion #104: Can't read mobileraker.conf in docker

ColdBeer72, 2024-12-03:
I defined my compose yml file for Mobileraker (using OpenMediaVault as host), as expected:

```yaml
services:
  mobileraker_companion:
    container_name: mobileraker
    environment:
      - PUID=${PUID}
      - PGID=${PGID}
      - TZ=${TZ}
    image: ghcr.io/clon1998/mobileraker_companion:latest
    volumes:
      - CHANGE_TO_COMPOSE_DATA_PATH/mobileraker/mobileraker.conf:/opt/printer_data/config/mobileraker.conf
      - /etc/hosts:/etc/hosts:ro
    restart: unless-stopped
```
But when I start it, these lines appears in log (just few starting lines):
```log
mobileraker  | 2024-12-03 22:21:37 root INFO Logging to file: /tmp/mobileraker.log
m`obileraker  | 2024-12-03 22:21:37 root INFO MobilerakerCompanion version: ?
mobileraker  | 2024-12-03 22:21:37 root INFO Checking if passed config exists: printer_data/config/mobileraker.conf
mobileraker  | 2024-12-03 22:21:37 root INFO Found configuration file at: /opt/printer_data/config/mobileraker.conf
mobileraker  | 2024-12-03 22:21:37 root INFO Read 1 printer config sections
mobileraker  | 2024-12-03 22:21:37 root INFO Main section read, language:"en", timezone:"CET", eta_format:"%d.%m.%Y, %H:%M:%S", include_snapshot:"True"
mobileraker  | 2024-12-03 22:21:37 py.warnings WARNING /opt/mobileraker/mobileraker.py:44: DeprecationWarning: There is no current event loop
mobileraker  |   event_loop = asyncio.get_event_loop()
mobileraker  | 
mobileraker  | 2024-12-03 22:21:37 mobileraker._Default INFO MobilerakerCompanion client created for _Default, it will ignore the following sensors: []
mobileraker  | 2024-12-03 22:21:37 mobileraker._Default.jrpc INFO Trying to connect to: ws://127.0.0.1:7125/websocket api key <NO API KEY>
mobileraker  | 2024-12-03 22:21:37 websockets.client INFO ! connect failed; reconnecting in 0.6 seconds
```
The config file defined in the yaml contains "language: es", correct moonraker hosts, and other things, so it's like if docker is trying to find the config file in another place. In fact, it generate a "Volume" in another place (like `'/var/lib/docker/volumes/508737591f2115acdf2dcad618bfeefc16a7997ee66d6765047cc82b4de4ce67/_data`' with a mobileraker.conf (0 bytes) inside.

If I edit and change that 0 bytes file with the contents of mine, mobileraker doesn't detect the changes, so it's never configured.
 
---
If I change the volumes to:
```yaml
      volumes:
      - CHANGE_TO_COMPOSE_DATA_PATH/mobileraker/config:/opt/printer_data/config
      - /etc/hosts:/etc/hosts:ro
```
Then log changes but can't find the configuration and says it can't find the configuration file.

---
After some more tests later, it seems as mobileraker (inside the dockerized vm) is not reading that file, or can't find it, or is reding another file.

Clon1998, 2024-12-03:
```docker
    volumes:
      - CHANGE_TO_COMPOSE_DATA_PATH/mobileraker/mobileraker.conf:/opt/printer_data/config/mobileraker.conf
```

You need to define the correct mapping for the mobileraker.conf file.

E.g. 
The file on the host machine is located at `/home/foo/mobileraker.conf` the mapping would look like:
```docker
    volumes:
      - /home/foo/mobileraker.conf:/opt/printer_data/config/mobileraker.conf
      ```

ColdBeer72, 2024-12-03:
> You need to define the correct mapping for the mobileraker.conf file.
> 
> E.g. The file on the host machine is located at `/home/foo/mobileraker.conf` the mapping would look like:

Oh, sure! You see `CHANGE_TO_COMPOSE_DATA_PATH` which is a variable (used in OpenMediaVault) to a data folder (at the moment it's translated on my machine as `/data/vmachines/appdata' and I could confirm to you the file exists:

[image]

and **[here](https://digistorage.net/brcinxwm)** you could donwload and check my mobileraker.conf file.

I've got 6 more docker-compose machines working in the same way on the same OpenMediaVault host.

Clon1998, 2024-12-03:
> Oh, sure! You see `CHANGE_TO_COMPOSE_DATA_PATH` which is a variable (used in OpenMediaVault) to a data folder (at the moment it's translated on my machine as `/data/vmachines/appdata' and I could [...]

Okay! I will have a look tomorrow to check what is going on!

Clon1998, 2024-12-04:
[image]

Hey, I had a look and I am unable to reproduce your issue. I tried to start the docker container via the cli with your config, and the container picks up the config just fine...

Clon1998, 2024-12-04:
I'd suggest you attach to the container and inspect the `mobileraker.conf` file in the container to make sure it is correctly transferred. You could also try to set the absolute path on your host instead of using `CHANGE_TO_COMPOSE_DATA_PATH` to see if the mapping is the reason.

ColdBeer72, 2024-12-04:
Solved!!!

The problem wasn't in the path style (nor the alias/variable) used. It works perfectly on this docker and others.

The problem came from user permissions.

By default, that file is writed on the host with user _root_ and group _libvirt-qemu_ and _chmod 660_. Even changing group to root for the file, mobileraker can't read it, so I changed the **read/write permissions** to **666** and then, it could be read by mobileraker.

As you see, dificult to find the origin, but you gave me a great idea attaching to the container's console. Thanks!!!

I think you could close the issue yet.

closed 2024-12-04 by Clon1998