Clon1998/mobileraker_companion #107: Error 'virtualenv: not found' during installation

FrankX0, 2025-01-25:
I encounter following error during installation:
```
...
System package install complete.
Checking Python Virtual Environment For Mobileraker Companion...
No virtual environment found, creating one now.
/usr/data/mobileraker_companion/scripts/install.sh: line 171: virtualenv: not found
```

Trying to install virtualenv by hand (`pip install virtualenv`) fails (pip not found).
Installing with pip3: `Requirement already satisfied: virtualenv in /usr/lib/python3.8/site-packages (15.1.0)`

Suggestions?

Clon1998, 2025-01-25:
Can you provide me more information about your host?
Is it just PiOS or a reality machine?

FrankX0, 2025-01-25:
A Creality K1.
I used the Creality-Helper-Script.

Clon1998, 2025-01-25:
> A Creality K1. I used the Creality-Helper-Script.

Ill verify it on my test K1 to see if creaility changed something.
However, I suggest you try to install it directly via the "Installer method" described in the readme.

FrankX0, 2025-01-25:
Thanks, but I was already executing the installer directly with the same result.

FrankX0, 2025-01-25:
I found the solution: `/usr/bin/easy_install virtualenv`
This installs virtualenv and allows the installation to finish.

TheNightmanCodeth, 2025-09-07:
FWIW, I was also able to get around this by changing the install.sh around line 170:

```
log_info "No virtual environment found, creating one now."
mkdir -p "${ENV_DIR}"
if [ "$IS_K1_OS" -eq 1 ]; then
    # The K1 requires we setup the virtualenv like this.
    python3 /usr/lib/python3.8/site-packages/virtualenv.py -p /usr/bin/python3 --system-site-packages "${ENV_DIR}"
else
    # Everything else can use this more modern style command.
    virtualenv -p /usr/bin/python3 --system-site-packages "${ENV_DIR}"
fi
```

it seems (perhaps on newer firmwares? really not sure) that `if [[ -f /opt/bin/python3 ]]; then` is unnecessary/wrong. There's a chance that when `/opt/bin/python3` does exist, `virtualenv` does not. 
Perhaps it's more appropriate to do `if [[ -f $(which virtualenv) ]]; then`. I really don't know, though.